rsd-devel/rsd #6: Add to LiteX ecosystem?

mithro, 2020-01-14:
RSD looks like it would be a super interesting addition CPU core option for the [LiteX Ecosystem](https://github.com/timvideos/litex-buildenv/wiki). LiteX already supports multiple different RISC-V cores (see https://github.com/enjoy-digital/litex/wiki/Soft-CPU for the list);
 - VexRISCV (https://github.com/SpinalHDL/VexRiscv)
 - Rocket
 - BlackParrot (https://github.com/black-parrot/pre-alpha-release)
 - others...

It would be awesome to see RSD in the LiteX system. It would let us very easily compare the resource usage of RSD to Black-Parrot, to Rocket, and VexRISCV in a fair environment.

[LiteX-BuildEnv](https://github.com/timvideos/litex-buildenv) is my environment which uses LiteX to create images for many different FPGA boards. It supports Zephyr, Linux, Micropython and bare metal firmware.

You can find a diagram of this below;
[image]

There is also a [Linux on LiteX VexRISCV](https://github.com/enjoy-digital/linux-on-litex-vexriscv) repository which would be cool to port to use RSD as an alternative option.

You can find the existing CPUs at https://github.com/enjoy-digital/litex/tree/master/litex/soc/cores/cpu
Rocket is probably the closest configuration which can be found at https://github.com/enjoy-digital/litex/blob/master/litex/soc/cores/cpu/rocket/core.py

mithro, 2020-01-14:
Related issues;
 - https://github.com/enjoy-digital/litex/issues/177 - Import the Taiga high performance RISC-V core
 - https://github.com/enjoy-digital/litex/issues/245 - Import the microwatt PowerPC core
 - https://github.com/enjoy-digital/litex/pull/278 - Pull request to import BlackParrot
 - https://github.com/enjoy-digital/litex/pull/186 - Experimental Support for 64-bit RocketChip

FYI Here are a collection of links related to Linux on RISC-V;
 * https://github.com/enjoy-digital/linux-on-litex-vexriscv
 * https://github.com/antmicro/litex-linux-readme
 * https://github.com/SpinalHDL/VexRiscv/issues/60 - Linux on VexRiscv
 * https://github.com/antmicro/litex-linux-riscv
 * https://github.com/antmicro/litex-rv32-linux-system
 * https://github.com/SymbiFlow/ideas/issues/35
 * https://github.com/SymbiFlow/ideas/issues/34

shioyadan, 2020-01-14:
Thank you for your attractive suggestions.  The LiteX ecosystem looks very good for usability, evaluation and maintenance for RSD. I'm not sure about the LiteX ecosystem, so I'd like to read the document and source code first, and then comment again.

mithro, 2020-01-14:
@shioyadan - I'm happy to provide advice and answer questions. The #litex IRC channel on irc.freenode.net is also a very useful resource. 

As actual performance is a lot more complicated than `DMIPS/Mhz` once you start taking into account things like memory subsystems and memory buses. I think the performance of your superscalar processor will be more interesting once those things are taken into account.