rsd-devel/rsd #15: flow to synthesis netlist and bitstream from Vivado alone

Daichou, 2020-04-08:
I find out that this project has a branch "support-vivado-alone-synthesis". Does this mean rsd is able to synthesis by Vivado only(no need Synplify)? 

In  Project/Vivado/Zedboard(support-vivado-alone-synthesis branch), it contains some vivado and xsdk script to generate vivado project. However, Project/Vivado/Zedboard/Script also contains some script to create vivado synthesis project. Whats the difference between two vivado post-synthesis scripts? In Project/Src/Makefiles/Vivado.inc.mk seems that it has rsd.vm (Synplify generate netlist) dependency.

reo-pon, 2020-04-08:
Yes, we are planning to be able to synthesize rsd by Vivado only, but we've not implemented it yet.

The scripts in Project/Vivado/Zedboard are used when synthesizing with netlist generated from Synplify (these scripts have existed in this repository since previous).
And we're going to add new script files to synthesis by Vivado only in Project/Vivado/Zedboard/Script.

I will add these files soon, so please wait for updates.

msmssm, 2020-05-07:
Hello Daichou,

We merged the "support-vivado-alone-synthesis" branch and now you can synthesis RSD with Vivado only.
The synthesis flow is described in the following wiki page.
https://github.com/rsd-devel/rsd/wiki/en-fpga-zynq-build-linux-simple

The post-synthesis simulation flow is here: https://github.com/rsd-devel/rsd/wiki/en-fpga-post-synth-simulation.

Thank you for your interest in our project!

closed 2020-05-07 by msmssm