rsd-devel/rsd #24: Use C libraries like string.h and stddef.h

odxa20, 2020-12-22:
Is there a way to use some of the libraries included with the gnu toolchain when writing programs ? Whenever I try to use the string library I get the following error

undefined reference to `strlen'

I have tried removing the -ffreestanding -fno-builtin -nostdlib	-nodefaultlibs flags from the Makefile but I've had no luck

shioyadan, 2020-12-30:
Sorry for my late reply.
In general, GLIBC integrated into GCC cannot be used in RSD because GLIBC works along with OS such as Linux. Instead of the default GLIBC, you have to use libc that can be used under a bare-metal environment without rich OS features.

We confirmed that some applications built for Zephyr OS, which is an embedded OS, can be executed on RSD. So, if you need libc features, my recommendation is to try Zephyr OS.

We are now working on providing some scripts that convert ELF binaries built for Zephyr OS into code.hex for RSD.

shioyadan, 2026-11-06:
It has been a very long time, but there has been some progress on this. The dev container on `develop` now includes picolibc. You can pass `LIBC=-lc` to your program's `make` command to link functions such as `strlen`.

I’ll close this issue now.

closed 2026-11-06 by shioyadan