rsd-devel/rsd #30: UnicodeDecodeError for RunTest.py script when running tests on Docker

sylwpro, 2021-01-25:
"UnicodeDecodeError: 'utf-8' codec can't decode byte 0xff in position 1: invalid start byte"

Files with serial data used for outStr, refStr in SerialOutputComparator class in script Processor\Tools\TestDriver\RunTest.py are opened in UTF-8 on Docker. Maybe it would be better if it was binary comparison.

shioyadan, 2021-01-29:
I confirmed that some test programs can actually output data that cannot be handled as UTF-8 and it can actually cause the problem. I am currently working on cleaning up test programs and I will change the script to treat data as binary in this branch. Thank you.

shioyadan, 2026-11-06:
Thank you for the report. This is fixed on `develop` in [25b39bc](https://github.com/rsd-devel/rsd/commit/25b39bc5823726caa81f9d6d67a0f6aa42b2d32c).