rsd-devel/rsd #31: Konata Wrong Statistics

odxa20, 2021-01-27:
I wanted to let you know of an issue I discovered regarding the Konata viewer. For a Bubblesort program I run (about 1000000 cycles) the hardware counters report 8926 branch prediction misses whereas the Konata statistics page reports 36 misses out of 2409 fetched branch instructions. I used the latest pre-built version of Konata. I am attaching the hex file and the source code of the bubble sort program and  screenshots of the verilator and Konata results
[image]
[image]
[bubblesort.zip](https://github.com/rsd-devel/rsd/files/5881182/bubblesort.zip)

shioyadan, 2021-01-29:
Thanks for reporting the issue. After investigating the issue, I found that it was a bug in Konata's statistics processing, and I fixed it. 

If you can run Konata from the latest source code, you will get correct statistics (the results will be slightly different from the numbers outputted from RSD because Konata cannot correctly count nested branch prediction misses). 

To launch Konata manually, you should install node.js and then "make init; make" (Please see README of Konata). I will release a new Konata binary soon.

odxa20, 2021-01-29:
Thank you very much. I will clone the Konata repository and test this as soon possible. Just to confirm. Even though nested branch mispredictions will not be counted the number of total branch instructions will be correct

closed 2021-02-01 by odxa20