rsd-devel/rsd #32: Issues while compiling the code

RaoShri, 2021-02-04:
Hey guys,
Sorry, I am new to this, so please help me out. I am trying to compile your code via verilator, I am getting the following errors:

%Warning-LATCH: LoadStoreUnit/StoreCommitter.sv:184:5: Latch inferred for signal 'dcWriteUncachable' (not all control paths of combinational always assign a value)
                                                     : ... In instance Main_Zynq_Wrapper.main.core
                                                     : ... Suggest use of always_latch for intentional latches
  184 |     always_comb begin
      |     ^~~~~~~~~~~
%Error: Exiting due to 18 warning(s)
make: *** [Makefile.verilator.mk:84: all] Error 1

Could you help me out in fixing this? Looks like issue from the design code.
This is the command I used : make -f Makefile.verilator.mk

msmssm, 2021-02-04:
Hello,

It is unable to compile RSD using older versions of Verilator.
If you are using Verilator version 3.x or older, please update it to 4.x or newer and try again.

Best Regards,
Susumu

RaoShri, 2021-02-04:
Hello,
I am currently using 4.108 version:
"Verilator 4.108 2021-01-10 rev v4.108-14-g2075db3d"

msmssm, 2021-02-04:
Hello,

It seems a compatibility problem of RSD with the latest verilator.
We will go over the problem soon, but please use Verilator 4.106 meanwhile.

RaoShri, 2021-02-04:
Hello Susumu,
I am able to compile on Verilator v4.106.
Thanks!
Regards,
Shrinidhi

closed 2021-02-08 by RaoShri