rsd-devel/rsd #44: Can RSD set the access permissions on a certain memory address?

Hsiang0816, 2021-10-03:
Hi,

I have seen the operation of RSD in memory mapping, and it seems to use simple static address translation instead of a mechanism like Page table. So is RSD unable to set memory permissions like the page table mechanism? Or does RSD have other methods to support memory permissions?

Thank you!

shioyadan, 2021-10-06:
Hi,

We have a plan to implement page protection in the future, but the current RSD only has a simple static protection mechanism. The simple mechanism does not allow you to dynamically change areas to be protected, but you can hardcode areas to be protected in HDL to provide fine-grained memory protection.

Hsiang0816, 2021-10-06:
Okay, Thank!

closed 2021-10-06 by Hsiang0816