rsd-devel/rsd #45: compile.tcl file is deprecated

RRozak, 2022-02-16:
[Processor/Project/DesignCompiler/compile.tcl](https://github.com/rsd-devel/rsd/blob/master/Processor/Project/DesignCompiler/compile.tcl) is deprecated.
There is file `CachePrimitives.sv` on the file list that is no longer in repository (removed in https://github.com/rsd-devel/rsd/commit/cca6d6e1cd4a863026ad6800b0dc277f58131f9c) and there are two files `HardwareCounter.sv` and `HardwareCounterIF.sv` that changed their names (in https://github.com/rsd-devel/rsd/commit/d77ca5306aed21bd265b8eccf6e537ee58ac8419)
I don't know if there are other problems.

shioyadan, 2022-02-17:
Thank you for your report.
I have confirmed that DesignCompiler/compile.tcl does not follow the current state of the source tree. I will make the necessary corrections in a few days.

By the way, I saw the related sv-tests issue. Does sv-tests extract the list of files needed to compile the RSD from DesignCompiler/compile.tcl?

shioyadan, 2022-03-29:
I'm sorry for the delay. Due to licensing issues with Design Compiler, I was unable to test my changes. 

I have merged a patch that fixes the problem into the master branch. Instead of enumerating the source code in the tcl file, the new version imports the source code list from another file. Although this method is more robust, it disables the python scripts in sv-tests. 

I am trying to fix the python script in sv-tests that you presented. I will issue a pull request when the fix is complete and would appreciate it if you could merge it.

Thank you.

shioyadan, 2022-03-30:
I have just submitted a PR that fixes the RSD generator in sv-tests to support the latest RSD. I hope the PR successfully fixes the issue.

https://github.com/chipsalliance/sv-tests/pull/2514

closed 2022-05-20 by shioyadan