rsd-devel/rsd #47: Does RSD support Supervisor mode or User mode?

Hsiang0816, 2022-03-30:
Hi!

After I read the documents and source codes of RSD, it seems that RSD can only support Machine mode.

Is it possible to support Supervisor mode or User mode? Or does RSD have technology like PMP etc...?

Thanks!

shioyadan, 2022-03-30:
Hi,

We would like to implement memory protection in the future. 
However, the current version of RSD supports only machine mode and does not support PMP.

RSD has its own simple static address translation mechanism. 
However, it is statically configured at synthesis time and may not be what you expect.
https://github.com/rsd-devel/rsd/blob/master/Processor/Src/Memory/MemoryMapTypes.sv

Hsiang0816, 2022-03-30:
Thank you for solving my doubts!

closed 2022-03-30 by Hsiang0816