rsd-devel/rsd #48: Benchmark access (RSD-env)

Dolu1990, 2022-04-27:
Hi, 

I was looking at some Coremark performance number to compare against https://github.com/SpinalHDL/NaxRiscv, but got multiple issues : 
- Can't find any documentation about coremark performances
- Can't find the Coremark port, as it is in the RSD-env closer repository

Would it be possible to either publish the coremark performances, or release some precompiled binaries, or open source a part of RSD-env ?

By the way, I used Konata quite a bit, it was very usefull, thanks ! :D 

Charles

shioyadan, 2022-05-02:
Hi,
I'm sorry for my late reply.

* Coremark/MHz in RSD is about 3.2. Recovery from branch misprediction in RSD is slow and thus it significantly reduces the total performance. I will try to improve the recovery mechanism.
* I have been preparing to release a Coremark port and binary for RSD, but it is not done yet. I haven't decided how to include Coremark in our repository (git sub-repository, fetching the repository in a Makefile, or including the source code as-is in our repository...).
* RSD-env contains some proprietary material that makes it difficult to publish.
* I noticed a while ago that your processor supports Konata, and I was very happy to see that! As you have probably noticed, Konata is very effective for debugging superscalar processors. If you have any feature requests, please don't hesitate to let me know.
* By the way, which log file format are you using, "kanata" or "gem5-O3PipeView"? The former is undocumented, but represents processor behavior in more detail (mouse over on konata to see more info) and is also used in RSD. Below is a google translation of the file format. I plan to document it in English.

https://github-com.translate.goog/onikiri/onikiri2/wiki/%E3%83%AD%E3%82%B0%E3%81%AE%E3%83%95%E3%82%A9%E3%83%BC%E3%83%9E%E3%83%83%E3%83%88?_x_tr_sl=ja&_x_tr_tl=en&_x_tr_hl=ja&_x_tr_pto=wapp

Thanks

Dolu1990, 2022-05-02:
Hi,
Thanks for the replay.

> Coremark/MHz in RSD is about 3.2. Recovery from branch misprediction in RSD is slow and thus it significantly reduces the total performance. I will try to improve the recovery mechanism.

Hooo, and coremark having those CRC being done with a if statement for each bit isn't realy being predictable, that's a real pain :/

> RSD-env contains some proprietary material that makes it difficult to publish.

Sure, i understand, real pain on my side too that many known benchmark aren't opensource themself (ex specint)
I used https://github.com/embench/embench-iot to avoid the issue a bit :)

> Konata is very effective for debugging superscalar processors. 

Yes, very helpfull, especialy tracking performance bug :D

> . If you have any feature requests, please don't hesitate to let me know.

Thanks :)

> By the way, which log file format are you using, "kanata" or "gem5-O3PipeView"? 

I used the "gem5-O3PipeView" one, as i was able to produce some references with gem3-03 itself XD

I was looking for the  "kanata" format, but as i wasn't finding example or a doc, i didn't tried much. But that would be something i would be interrested in, as it would provide more details.

Thanks ^^
Charles

shioyadan, 2026-11-06:
Hi,

It has been a very long time, but I wanted to share an update. I recently started overhauling RSD on the `develop` branch, using AI to help with the cleanup and improvements.

As part of this work, I have updated the setup for building and running benchmarks such as CoreMark. The CoreMark port now builds from public sources without `RSD-env`. The [build and run instructions](https://github.com/rsd-devel/rsd/blob/develop/Processor/Src/Verification/TestCode/README.md#coremark) are available.

Once this overhaul is complete, I plan to merge further work, including the MMU implementation branch, with the goal of booting Linux.

I’ll close this issue now. Thank you again for your interest in RSD and for our earlier discussions!

closed 2026-11-06 by shioyadan