rsd-devel/rsd #64: error running the Makefile.verilator.mk

Dennishor123, 2023-04-03:
im trying to run make -f Makefile.verilator.mk run and i got this error:
../Project/Verilator/obj_dir/VMain_Zynq_Wrapper \
	MAX_TEST_CYCLES=100000 \
	TEST_CODE=Verification/TestCode/Asm/FP ENABLE_PC_GOAL=1 SHOW_SERIAL_OUT=1
make: ../Project/Verilator/obj_dir/VMain_Zynq_Wrapper: No such file or directory
make: *** [Makefile.verilator.mk:93: run] Error 127
 im wondering where does the /Project/Verilator/obj_dir/VMain_Zynq_Wrapper comes from as i couldnt find it in this github, thanks a lot

shioyadan, 2023-04-05:
You should execute "make -f Makefile.verilator" for compilation before "make -f Makefile.verilator run." 
The build and simulation example in the README was broken and has been fixed. Sorry for the inconvenience.

aignacio, 2024-06-22:
Hey @shioyadan,

I just tried the below cmd and I'm having the following issue while running with Verilator:
```bash
 docker run -it --rm -v ./:/test/ -w /test/ aignacio/rtldev bash -c 'cd Processor/Src && make -f Makefile.verilator.mk'
```
[image]

shioyadan, 2024-06-27:
Hello,

I'm not entirely sure what the specific issue is, but it seems to relate to a known problem in newer versions of Verilator, as detailed in the following discussion:
https://github.com/verilator/verilator/issues/5139

This issue has been reported multiple times, and you may be able to resolve it by downgrading to an older version from the 4.x series of Verilator.

shioyadan, 2026-11-06:
It has been a long time, but I wanted to share an update: `develop` now supports Verilator 5, and the dev container provides a tested version. Please follow the updated [quick start](https://github.com/rsd-devel/rsd/blob/develop/README.md#getting-started). We are closing this older setup issue; please let us know if you encounter problems with the current environment. Thank you for your report.

closed 2026-11-06 by shioyadan