rsd-devel/rsd #67: cannot convert ‘const VlWide<92>’ to ‘const uint32_t (&)[92]’ {aka ‘const unsigned int (&)[92]’}

Chenwei-user, 2023-05-15:
I use the verilator branch v4.026-2-g0c6c83e to run Simulation. But when I make Makefile.verilator.mk, I get the error
"In file included from ../../../Src/SysDeps/Verilator/TestMain.cpp:4:
../../../Src/SysDeps/Verilator/VerilatorHelper.h:475:85: error: cannot convert ‘const VlWide<92>’ to ‘const uint32_t (&)[92]’ {aka ‘const unsigned int (&)[92]’}
  475 |         d->memberName0.memberName1 = h->DebugRegister_##memberName0##_##memberName1(r); \
      |                                                                                     ^
      |                                                                                     |
      |                                                                                     const VlWide<92>
../../../Src/SysDeps/Verilator/VerilatorHelper.h:732:5: note: in expansion of macro ‘RSD_MAKE_STRUCT_ACCESSOR_LV2’
  732 |     RSD_MAKE_STRUCT_ACCESSOR_LV2(DebugRegister, perfCounter, DataPath, numBranchPredMissDetectedOnDecode)
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from ./VMain_Zynq_Wrapper__Syms.h:16,
                 from ../../../Src/SysDeps/Verilator/TestMain.cpp:2:"
It seems like a verilator version problem, How should I fix it?

Chenwei-user, 2023-05-16:
I have address it. Beacause I don't set the right VERILATOR_ROOT besides RSD_VERILATOR_BIN

closed 2023-05-16 by Chenwei-user

zhangkanqi, 2023-12-07:
How did you fix this error? I also encountered this problem. #72

Chenwei-user, 2023-12-07:
Add an Environment Virable: VERILATOR_ROOT, and set it as same as RSD_VERILATOR_BIN

Chenwei-user, 2023-12-07:
Get it!, Problem is the version:  Verilator 4.026 2020-01-11 rev v4.026-2-g0c6c83e is required. Check test environment carefully.

Chenwei-user, 2023-12-07:
Verilator 4.026 is not enough, branch v4.026-2-g0c6c83e is needed.

zhangkanqi, 2023-12-07:
why?
Have you tried this before?

Chenwei-user, 2023-12-07:
Yes, I have try it. For more information, you can email me.:chenwei_cs@stu.pku.edu.cn