rsd-devel/rsd #70: wrong pointer used for poping btb queue

robert-g-liu, 2023-08-15:
Literally in the btbQueuePointer the comment pointes out _The "tailPtr" points the address of a next pushed entry._

BTB used **headPtr** for pushing entry. See code line below.

>         else if (pushBtbQueue) begin
>             btbQueue[headPtr].btbWA <= btbWA[INT_ISSUE_WIDTH-1];
>             btbQueue[headPtr].btbWV <= btbWV[INT_ISSUE_WIDTH-1];
>         end 

Similarly, to pop the queue, **tailPtr** was used so stale/invalid data was fetched:

>         // Pop btb Queue
>         if (!empty && !updateBtb) begin
>             popBtbQueue = TRUE;
>             btbWE[0] = TRUE;
>             btbWA[0] = btbQueue[tailPtr].btbWA;
>             btbWV[0] = btbQueue[tailPtr].btbWV;
>         end

robert-g-liu, 2023-08-15:
Gshare branch predicator has the same issue. After local fix, improvement on both Num of branch prediction misses and IPC could be seen.

reo-pon, 2023-08-29:
Thank you for reporting the issue.
What you pointed out is indeed correct.
I have confirmed that the number of branch mispredictions and IPC improved after the fix.

I will push the fix to the repository soon.

shioyadan, 2026-11-06:
Thank you for the report. This is fixed on `develop` in [7d64438](https://github.com/rsd-devel/rsd/commit/7d64438a12bb51500783939a63c2b012b1405481).

closed 2026-11-06 by shioyadan