rsd-devel/rsd #97: [Bug Report] RSD is excuting illegal instruction

chenhychen, 2024-11-01:
Hi,

I have found that RSD can execute illegal instruction in the commit `7b65f6b`.
According to RISC-V Instruction Set Privileged Architecture, the `funct3` of the load/store instruction should be from `000` to `110`.
However, if the `funct3` is set to `111`, the RSD can still execute the instruction but Spike throws the illegal instruction exception.

For example, executing the instruction hex code `0x00817703` will cause Spike to throw the illegal instruction exception, but RSD executes it. Attach a hex file that can trigger this vulnerability.

[illegal_load.zip](https://github.com/user-attachments/files/17602550/illegal_load.zip)

reo-pon, 2025-01-27:
Hi. I apologize for my late reply.
Thank you for reporting the bug.

As you pointed out, the instruction `7b65f6b` should cause an invalid instruction exception, but RSD processes it as a valid instruction.
I will fix this bug soon.

shioyadan, 2026-11-06:
Thank you for the report. This is fixed on `develop` in [70c4de9](https://github.com/rsd-devel/rsd/commit/70c4de9c920dc527e6d4af1cb354da90ca71cd88).

closed 2026-11-06 by shioyadan