rsd-devel/rsd #99: Error running RTL simulaiton

mutianzh, 2025-02-14:
Hi,

I'm trying to use Questasim to run RTL simulation. If I run command 'make' under Src, I'll get the following error:

Errors: 0, Warnings: 0
/tool/mentor/questa_ultra_10.7b/questasim/bin/vsim -c    ../Project/ModelSim/Main.TestMain  -do "quit" # link check
Reading pref.tcl

# 10.7b

# vsim -c ../Project/ModelSim/Main.TestMain -do "quit" 
# Start time: 13:34:19 on Feb 12,2025
# ** Note: (vsim-3812) Design is being optimized...
# ** Note: (vopt-143) Recognized 1 FSM in module "CacheFlushManager(fast)".
# ** Error (suppressible): ./Cache/DCache.sv(178): (vopt-2577) Enum type mismatch between operands of '!=' expression.
# ** Error: (vopt-2064) Compiler back-end code generation process terminated with code 2.
# ** Note: (vopt-143) Recognized 1 FSM in module "QuickDivider(fast)".
# ** Note: (vopt-143) Recognized 1 FSM in module "RenameLogicCommitter(fast)".
# ** Note: (vopt-143) Recognized 1 FSM in module "RenameStageSerializer(fast)".
# ** Error (suppressible): ./Cache/DCache.sv(515): (vopt-2577) Enum type mismatch between operands of '>=' expression.
# ** Note: (vopt-143) Recognized 1 FSM in module "ICache(fast)".
# ** Error: (vopt-2064) Compiler back-end code generation process terminated with code 2.
# ** Error (suppressible): ./LoadStoreUnit/StoreQueue.sv(289): (vopt-2577) Enum type mismatch between operands of '!=' expression.
# ** Note: (vopt-143) Recognized 1 FSM in module "FP32DivSqrter(fast)".
# ** Error: (vopt-2064) Compiler back-end code generation process terminated with code 2.
# Error loading design
Error loading design
# End time: 13:34:21 on Feb 12,2025, Elapsed time: 0:00:02
# Errors: 6, Warnings: 0, Suppressed Errors: 166
make: *** [Makefile.inc:186: rtl] Error 12

I appreciate any guidence

reo-pon, 2025-02-17:
Hi.

It seems likely that the error is occurring because you are using an older version of QuestaSim. The syntax causing the error is supported in newer versions.

I have confirmed that RSD can be successfully simulated with versions 2019.4_2 and 2023.1_1. I think it should be possible to simulate without any problems after 2019.4_2.

You may be able to resolve this issue by trying one of the following:

- **Use a newer version of QuestaSim:** If you have access to different versions, I recommend using 2019.4_2.

- **Suppress the error with compilation options:** Based on your error log, modifying the Makefile should allow you to suppress the error. However, keep in mind that differences between QuestaSim versions might lead to unexpected behavior during simulation.

```
diff --git a/Processor/Src/Makefile b/Processor/Src/Makefile
index 7f1b1d9..7a6f321 100755
--- a/Processor/Src/Makefile
+++ b/Processor/Src/Makefile
@@ -46,6 +46,7 @@ VSIM_OPTIONS = \
        +ENABLE_PC_GOAL=$(ENABLE_PC_GOAL) \
        +SHOW_SERIAL_OUT=$(SHOW_SERIAL_OUT) \
        -suppress 8233 \
+       -suppress 2577 \
        +autofindloop \
```

If you have any problems or questions, please feel free to ask.

mutianzh, 2025-02-17:
Hi,

I can now run simulations by suppressing the error. Many thanks!
I would still get the same error log by just running 'make' because the link check command does not include any VSIM_OPTIONS, but I can still run simulaiton after that.

mutianzh, 2025-02-17:
Hi,
I can now run simulations after supressing the error. Many thanks!
I will still get the same error log by just using 'make' because the link check command doesn't use any VSIM_OPTIONS. I can still run simulation after that.

closed 2025-02-17 by mutianzh